ttlappalainen/NMEA2000_mcp #2: SREG not defined?

chr-be, 2016-11-12:
Hi, greate Project!
While trying the libs for an ESP8266 i got an Error, that SREG is not defined in NMEA2000_mcp.ccp.

After defining uint8_t SREG; the compiler fond no errors and the connectin starts. But i´, not so deep in it, to understand, what SREG is realy for....
Using Arduino 1.6.12
Best Regards,
Christian

ttlappalainen, 2016-11-17:
Sorry for delay - I forgot to press "comment" button.

What board you are using for base? SREG is used for interrupts state and restoring it after critical area. There are some discussion about nested state changes so I used succested one. All platforms does not have SREG and that is causing problem

alex-pole, 2018-02-17:
Hei Timo!
Got the same when tried to compile for Due. Unfortunately SREG is not supported there since Due is ARM-based.
I want to use Due with MCP2515+TJA1050 because I already have small board with them (and it's quite difficult to get mcp2562). If you can guide how to fix it I can do it and send pull request. If it is complicated or not required I will try to get mcp2562.
Thank you!!

ttlappalainen, 2018-02-18:
I do not see any good points to use mcp_can with DUE. At least Farnell and Digikey have MCP2652 in stock.

Just now I do not have time to think best fix.

alex-pole, 2018-02-24:
You are right. Got SN65HVD234 and made it working with Due! 

I'm using Due as a nmea traffic generator and connected it to RPi over socketCan (by mcp2515 on RPi side). SocketCan port of NMEA_can also works fine. I use RPi to send traffic over IP (nmea UDP datagrams over wlan), tested with OpenSkipper, so the whole chain (Due-RPi-OpenSkipper) works fine. 
Final goal is to send traffic from the boat to RPi and then to Expedition Marine software (most popular racing software) via wlan. Expedition accepts also Activision data so it is a workaround if I fail with IP or Expedition is using some custom message format.

Thank you very much for the code. Greetings from Pietari!

PS Do you have any clue on OpenSkipper project? Is it still alive?

ttlappalainen, 2018-02-24:
Nice to hear you got it running. 

OpenSkipper is alive. You will find sources now on https://github.com/OpenSkipper/OpenSkipper/ There is not development yet, but there is new person working with it.

Have you checked my NMEA simulator for generating test traffic? You need PC, DUE and NMAE2000 library ActisenseListenerSender example to feed data from PC to the bus.

alex-pole, 2018-02-25:
I generate traffic with something similar to your Examples/MessageSender. Still Due generates not that huge load, need to play around with it to boost, I will do it later.

1. As of now I'm trying to connect RPi via IP (UPD over WiFi) to Expedition software I mentioned and the problem is that it accepts traffic only from particular devices and I believe mostly all of them send data in proprietary formats. I don't have any of these devices in hardware so unable to reverse it so just want to check with you if might know message format these devices generate when communicating over IP? As for format OpenSkipper is using I just was lucky enough to guess it. :)
Devices are: 
A+T 
B&G H3000/F15000 HLink 
B&G H3000/F15000 Hydra 
B&G H390 
B&G HSO00 WebSocket 
B&G H690/F12000 Perf 
B&G Hydra 
Bravo Systems 
Chetco SeaSmart 
Cosworth 
DMK Yacht 
Expedition 
Garmin NX2 FDX 
Garmin NX2 NMEA 
KND iDataNet 
KND Onboard Assistant 
Koden MDS-1/8/9/10R 
Koden MDS-5/6R 
KVH Quadro 
Laser
Maretron SSC200 
NKE 
Ockam 
Sailmon 
Stowe 
TackTick 
Todter 
Vesper
VSPARS
Yacht Devices

2. I'm using RPi to send data generally because it has built in WiFi adapter. Another option might be Due itself with some sort of ESP8266 to send data over WiFi directly. Not sure if UDP multicast is working there properly, need to check. They sell such kind of devices (NMEA over WiFi) for about 200 euro which is definitely not a fair price and bringing such DIY solutions to community is a great thing!

PS Is there any way to communicate with you directly? It is not a great idea to discuss here topics which are irrelevant to the issue.

ttlappalainen, 2018-02-26:
1. I think there is no standard. The ones I have seen uses just NMEA0183 and some sends NMEA2000 messages in SeaSmart format. Search "seasmart protocol". There is also seasmart module included in the library for converting tN2kMsg to seasmart.

2. I prefer Teensy based boards. They use less power and has more flash/ram. SN65HVD234 works with them too.

Find contact via kave.fi