ttlappalainen/NMEA2000_mcp #3: problems for ESP8266 and MCP2515

jopimo, 2018-10-02:
Hello,

when I try to compile for ESP8266 get 

In file included from /Users/master/Documents/Arduino/libraries/NMEA2000/src/NMEA2000_CAN.h:144:0,
                 from /var/folders/p_/j24qpcws45d_0kzx14mtvdrm0000gn/T/arduino_modified_sketch_357838/BatteryMonitor.ino:15:
/Users/master/Documents/Arduino/libraries/NMEA2000_mcp/NMEA2000_mcp.h: In member function 'bool tNMEA2000_mcp::tFrameBuffer::AddFrame(long unsigned int, unsigned char, const unsigned char*) volatile':
/Users/master/Documents/Arduino/libraries/NMEA2000_mcp/NMEA2000_mcp.h:98:20: error: no matching function for call to 'min(unsigned char&, int)'
       len=min(len,8);
                    ^
/Users/master/Documents/Arduino/libraries/NMEA2000_mcp/NMEA2000_mcp.h:98:20: note: candidates are:
In file included from /Users/master/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/algorithm:62:0,
                 from /Users/master/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/Arduino.h:254,
                 from /var/folders/p_/j24qpcws45d_0kzx14mtvdrm0000gn/T/arduino_modified_sketch_357838/BatteryMonitor.ino:11:
/Users/master/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algo.h:4226:5: note: template<class _Tp, class _Compare> _Tp std::min(std::initializer_list<_Tp>, _Compare)
     min(initializer_list<_Tp> __l, _Compare __comp)
     ^
/Users/master/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algo.h:4226:5: note:   template argument deduction/substitution failed:
In file included from /Users/master/Documents/Arduino/libraries/NMEA2000/src/NMEA2000_CAN.h:144:0,
                 from /var/folders/p_/j24qpcws45d_0kzx14mtvdrm0000gn/T/arduino_modified_sketch_357838/BatteryMonitor.ino:15:
/Users/master/Documents/Arduino/libraries/NMEA2000_mcp/NMEA2000_mcp.h:98:20: note:   mismatched types 'std::initializer_list<_Tp>' and 'unsigned char'
       len=min(len,8);
                    ^

ttlappalainen, 2018-10-03:
That is because inaccurate definitions and since ESP8266 environment does not define min as arduinos (which is actually stupid).  Change to code
len=(len<8?len:8);
And it should compile. I'll fix it to the code git later.

There is also other bug found on mcp_can, which prevent system sending anything. I'll try to fix it today.

ttlappalainen, 2018-10-04:
Now I updated both NMEA2000_mcp and mcp_can.

jopimo, 2018-10-07:
Ok no more robles with min, but....

( trying to compile the example battery monitor,    Thank you very much. ) 

/Users/master/Documents/Arduino/libraries/NMEA2000_mcp/NMEA2000_mcp.cpp: In member function 'virtual bool tNMEA2000_mcp::CANSendFrame(long unsigned int, unsigned char, const unsigned char*, bool)':
/Users/master/Documents/Arduino/libraries/NMEA2000_mcp/NMEA2000_mcp.cpp:98:24: error: 'SREG' was not declared in this scope
     uint8_t SaveSREG = SREG;   // save interrupt flag
                        ^
/Users/master/Documents/Arduino/libraries/NMEA2000_mcp/NMEA2000_mcp.cpp: In member function 'virtual bool tNMEA2000_mcp::CANGetFrame(long unsigned int&, unsigned char&, unsigned char*)':
/Users/master/Documents/Arduino/libraries/NMEA2000_mcp/NMEA2000_mcp.cpp:167:26: error: 'SREG' was not declared in this scope
       uint8_t SaveSREG = SREG;   // save interrupt flag
                          ^
/Users/master/Documents/Arduino/libraries/NMEA2000_mcp/NMEA2000_mcp.cpp: In member function 'void tNMEA2000_mcp::InterruptHandler()':
/Users/master/Documents/Arduino/libraries/NMEA2000_mcp/NMEA2000_mcp.cpp:209:83: error: invalid conversion from 'volatile uint32_t* {aka volatile unsigned int*}' to 'volatile long unsigned int*' [-fpermissive]
         N2kCAN.readMsgBufID(status,&(frame->id),&ext,&rtr,&(frame->len),frame->buf);
                                                                                   ^
In file included from /Users/master/Documents/Arduino/libraries/NMEA2000_mcp/NMEA2000_mcp.h:33:0,
                 from /Users/master/Documents/Arduino/libraries/NMEA2000_mcp/NMEA2000_mcp.cpp:24:
/Users/master/Documents/Arduino/libraries/CAN_BUS_Shield/mcp_can.h:138:10: error:   initializing argument 2 of 'byte MCP_CAN::readMsgBufID(byte, volatile long unsigned int*, volatile byte*, volatile byte*, volatile byte*, volatile byte*)' [-fpermissive]
     byte readMsgBufID(byte status, volatile unsigned long *id, volatile byte *ext, volatile byte *rtr, volatile byte *len, volatile byte *buf); // read buf with object ID
          ^
exit status 1
Error compilando para la tarjeta NodeMCU 1.0 (ESP-12E Module).