ttlappalainen/NMEA2000_mcp #4: Arduino MKR Zero & MKR Can Shield -> SREG not defined, cli(), sei()

christoph12345, 2018-10-18:
I use the Arduino MKR Zero (SAMD21 Cortex-M0+ based) and the MKR Can shield (MCP2515 based) with your librarys.
The hardware works, I can read NMEA2000 packets interpreted as standard can extended packets using a can library.

I used the Actisense Listener example provided with the library, and only put these lines before including anything for configuring the MCP2515 correctly:
#define USE_N2K_CAN 1
#define N2k_SPI_CS_PIN 3
#define N2k_CAN_INT_PIN 7
#define USE_MCP_CAN_CLOCK_SET 16

The compiler was not able to compile NMEA2000_mcp.cpp for the ARM, because it has no SREG, so I commented it out. I also had to replace cli() and sei() with noInterrupts() and interrupts().
Then I was able to compile it, but I did not receive any messages. I think the changes I made are not complete.

Has anyone tried it with the MKR Zero and CAN Shield yet? I think this would be the perfect hardware.

ttlappalainen, 2018-10-18:
I still like Teensy more...

The changes should be OK with quick look. I should add defines for NMEA2000_mcp.cpp - others has also used mcp_can on different environment.

Do you use mcp_can library from my git? I am not sure of status from other sources and mine can also handle both transmit and receive with interrupt. You could also try to leave int pin undefined, so it should use just polling, which is enough on the tests.

I have not heard others using MKR.

christoph12345, 2018-10-19:
I have to use the MKR Zero, because I do not built the electronics from scratch, I just add NMEA2000 functionality to an already existing project...
Yes I use your mcp_can library.
I will try leaving the int pin undefined in the evening.

ralfhille, 2019-04-11:
Is this issue solved? I want to use the Arduino MKR WIFI 1010 and the MKR CAN shield.
My current solution works with MEGA and CAN shield from Seed, but i want to chnage to smaller Hardware.

ttlappalainen, 2019-04-12:
Should be. I have modified NMEA2000_mcp to use SREG only with Mega and Uno.

All Teensy 3.2 and up has internal can controller and requires tranceiver ship. I have used MCP2562 without any problem. With ISO1050 tranceiver you can make isolated system.

michydk, 2022-01-12:
As a feedback to ttlappalainen, and to those that find this tread in the future:
I have with success, used a Arduino MKR Wi-Fi 1010 and the MKR CAN shield on a NMEA2000 network, using ttlappalainen's NMEA2000 library and a Raymarine axiom 7 as MFD

I have 3 ds18b20 (one-wire Temperatur sensor) connected on a one-wire bus, then senses temperatures on my old VP 2003t engine.

For testting purpose, the alarm is set to 26 C, in case you wondered..;)

I can highly recommend ttlappalainen's excellent document "Connecting_hardware_to_NMEA2000.pdf", specially the troubleshooting section :
[image]
[image]
)

oskarbob, 2024-07-23:
Cannot find a trace of "Connecting_hardware_to_NMEA2000.pdf" anywhere.   Where did you see this?

ttlappalainen, 2024-07-23:
It is old document and moved as part of NMEA2000 library document: https://ttlappalainen.github.io/NMEA2000/page_h_w_set_up.html

oskarbob, 2024-07-23:
Thank you.

Links:
------
[1] 
https://linkprotect.cudasvc.com/url?a=https%3a%2f%2fttlappalainen.github.io%2fNMEA2000%2fpage_h_w_set_up.html&amp;c=E,1,_99QL2EMbOHFlHE8O8KQa65x5Wp02U4Yu9FrIRjWC6yatu28S17DtoG8UZV-iK1-papfqaoYsj2aKEQvGv629lz_B2refQ2J22drwBPS8pFNy1c,&amp;typo=1
[2] 
https://github.com/ttlappalainen/NMEA2000_mcp/issues/4#issuecomment-2244128762
[3] 
https://github.com/