ttlappalainen/NMEA2000_mcp #8: MCP_CAN library receive problem - simple solution

hellmutbrudny, 2023-02-28:
Hi. I've just been testing Longan CANBed 2040 board with integrated mcp2515. It didn't worked because https://github.com/ttlappalainen/CAN_BUS_Shield/blob/master/mcp_can.cpp -> MCP_CAN::mcp2515_read_canMsg() lacks SPI_HAS_TRANSACTION handling. All need to be done is adding SPI_BEGIN/END around:

`#ifdef SPI_HAS_TRANSACTION
    SPI_BEGIN();
#endif
  MCP2515_SELECT();
...
#ifdef SPI_HAS_TRANSACTION
    SPI_END();
#endif
MCP2515_UNSELECT();`

Apparently original SeedStudion library has the same problem. But it is specifically for their board while MCP_CAN probably can handle any board with mcp2515 (I have been testing it with Arduino Nano Every + cheap MCP2515 CAN module as well as with RPi Pico + Waveshare RS485 CAN HAT).

And by the way - big thank you @ttlappalainen for the great job! It gave me a lot of inspiration and help.
I hope my comment will help somebody too ;).

ttlappalainen, 2023-02-28:
I do not use mcp_can anymore. Do you have any idea will that effect to other boards used with mcp_can?

hellmutbrudny, 2023-02-28:
I guess it will affect any MCU board that needs/uses SPI transactions (to communicate with mcp2515) - on plus ;). But since it's under compiler condition which is widely used in this file I think it's rather safe. If that's what concerns you?